davidtgbe/Marlin #2: Printing from SD Card fails

rizwansarwar, 2020-07-27:
While printing from SD Card, print stops randomly in the middle of a print causing the firmware to crash and reboot the printer. I have tried this various GCodes and slicers with same result. Printing from USB does not suffer from this issue and I have had good 12+ hours prints.

dozer123456, 2020-07-27:
I can confirm the same. Have tried multiple slicers and SD cards.

davidtgbe, 2020-07-28:
I think slow format and high quality SD could help. You should get same results formatting through diskpart or regular windows format. Refreshing SD sectors performing a slow format may improve printings, but, it is not a reliable solution. SD card reader is too sensitive or unstable.
It needs a solution. Among things that could be related with these instabilities are clock issues, so, bus clock config is a good point to start.

davidtgbe, 2020-07-28:
I've just commited changes with new bus clock configuration. 
If nothing improves, try to uncomment (#-DTRANSFER_CLOCK_DIV=8) on platform.io.

davidtgbe, 2020-07-28:
Printed cube with no problems. Waiting for feedback.

dozer123456, 2020-07-28:
Thanks David

Just compiling now will try printing a benchy as every time I try that with
the anet SD it fails within 10 min.

Will keep you updated.

dozer123456, 2020-07-28:
Failed again but this time I made it to around 75%😫

I will uncomment the line in the plstformio.ini and try again.

Moody66, 2020-07-28:
You mean the ANET sd card? That thing is JUNK. Literally. OR did you men the SD slot?

dozer123456, 2020-07-28:
Anet provided SD card 😂
And yes it probably is junk, but it worked ok-ish with the original
firmware so it probably is a good card to use to pick up any issues.

Moody66, 2020-07-28:
I had MAJOR issues using it. First thing I did was trash it. LOL I will flash soon as I ma done working and use my Samsung.

dozer123456, 2020-07-28:
Failed again at around the same point about 5 layers earlier this time. this is with the #DTRANSFER_CLOCK_DIV=8 uncommented DTRANSFER_CLOCK_DIV=8.

I Will try now with my Sandisk SD

dozer123456, 2020-07-28:
Partial Success.
The benchy print finished this time, but I still got a lot of strange moves, where the printhead would leave the print then return leaving loops of filament. 
The same gcode over USB prints perfect.

Moody66, 2020-07-29:
His change looks correct to me. I base this on these two pages.... 

https://stm32f4-discovery.net/2015/01/properly-set-clock-speed-stm32f4xx-devices/
And 
https://community.arm.com/developer/tools-software/tools/f/keil-forum/32994/changing-clock-and-pll-setting

dozer123456, 2020-07-29:
Based on that guide pll_m should=16

Bare in mind I don't know what I am taking about 🤣

The settings did improve things a great deal. But like I said. Printing on
a freshly formatted SanDisk SD I got a couple of funny print moves that
left some funny strings outside of the print. With the same SD card plugged
into the pc and loaded into cura, printing over usb gives a perfect result
👍

davidtgbe, 2020-07-29:
@dozer123456, first of all, thank you for your tests!
pll_m has to be set to 8 because we are using external clock @8Mhz.
It seems that SDIO driver on STM32F4 is not well tested, and might cause problems:
https://github.com/davidtgbe/Marlin/blob/f5830b82380421175145fa197654447c01cac71f/Marlin/src/HAL/STM32/Sd2Card_sdio_stm32duino.cpp#L89

dozer123456, 2020-07-29:
@davidtgbe I really appreciate all the effort you have gone to to get this working.

Completing test prints is the easy part 👍

Moody66, 2020-07-29:
> @dozer123456, first of all, thank you for your tests!
> pll_m has to be set to 8 because we are using external clock @8Mhz.
> It seems that SDIO driver on STM32 HAL is not well tested, and might cause problems:
> https://github.com/davidtgbe/Marlin/blob/f5830b82380421175145fa197654447c01cac71f/Marlin/src/HAL/STM32/Sd2Card_sdio_stm32duino.cpp#L89

Has Anet been helping at all?

davidtgbe, 2020-07-29:
I have changed from SDIO to SPI. It's not the best solution, but, I think it is the fastest way to solve the problem.
@Moody66 Latest conversation with Anet was last week. They kindly provided powerloss identification and were going to take a look to the bootloader.

anigmad, 2020-07-29:
Nice. Glad they are helping.  Boot loader will be nice.

Moody66, 2020-07-29:
Got nothing to do at the moment. All Networks are purring along. LOL. IF it is ready to test I will compile and try it.

davidtgbe, 2020-07-29:
@Moody66 yes, it is ready. Go ahead

dozer123456, 2020-07-29:
@davidtgbe  We have a perfect print. 😊
@Moody66  With the Anet SD card 🤣

Top job!!!!!!!!!

dozer123456, 2020-07-29:
Oh, by the way, the Filament runout detector works great.

I will be testing the power loss shortly.

👍

1uknowleast, 2020-07-29:
I have a rebadged Anet ET4 from LABISTS, with a ET4 1.1 board, so i'm watching the development with great interest. Great Work BTW!

Anecdotally I've seen the print head make random moves during an SD Card print, using the provided Sd Card (Samsung 8GB) and pre sliced Dog STL file. I checked the gcode online, it doesn't show that any random moves creating any funny strings.

This in on the LABISTS V1.0 firmware, so it maybe its not Marlin. Just my 2 cents.

All my other files via GCODE generated by Prusaslicer have been without issue.

dozer123456, 2020-07-29:
Power loss did recovery didn't work.

and I had enabled it with the m413 sn 🤣

Do I need to uncomment and change //#define POWER_LOSS_PIN         44  to PA8 ?

I can see the POWER_LOSS_PIN has been set to PA8 in the pins_ET4

davidtgbe, 2020-07-29:
@dozer123456 Great. One thing less in the to-do list. 
Regarding powerloss, one definition is enough. I haven't tested power loss, but, if you define power_loss_state = high instead of low (current value) , you will get a power outage error screen as soon as you start printing, so, I thought it would work...

@1uknowleast could you try building latest source code? This issue was created in part for the problems you mention.

1uknowleast, 2020-07-29:
> @1uknowleast labists? First time I hear. May be only a rebranding?
> Anyway, those movements seem to be related with sdcard. People printing from usb are not suffering them. If you have seen them on an Anet based FW, may be the have the same problem, or may be it is a bad SD...

Yes they are fairly new 3d printer maker, mostly Raspberry and Ardunio kits before, definitely a rebranding. I've only had this printer a few weeks and printed about a dozen or so items. Some via Octoprint, and some via the SD card. Only seen the issue once with the presliced gcode. However I have formatted the SD card since, so who knows.

dozer123456, 2020-07-30:
I have been doing testing.
on power loss, a PLR file is written to the SD but it is BLANK 0 bytes.

If I change the power loss pin state to HIGH, as soon as you start the print it stops with a power loss warning message.
this will write aPLR file to the SD card, but this time it holds meaning full data.

Do you think that changing the sd card to SPI it is now not fast enough to write the file when there is a power loss?

davidtgbe, 2020-07-31:
Seems to be fixed using SPI

closed 2020-07-31 by davidtgbe