davidtgbe/Marlin #3: Power loss

davidtgbe, 2020-07-30:
I have been doing testing.
on power loss, a PLR file is written to the SD but it is BLANK 0 bytes.

If I change the power loss pin state to HIGH, as soon as you start the print it stops with a power loss warning message.
this will write aPLR file to the SD card, but this time it holds meaning full data.

Do you think that changing the sd card to SPI it is now not fast enough to write the file when there is a power loss?

_Originally posted by @dozer123456 in https://github.com/davidtgbe/Marlin/issues/2#issuecomment-666285234_

davidtgbe, 2020-07-30:
@dozer123456 could be. I would have to debug the code to check, looks like what you have suggested. Don't know if it's a matter of speed because of SPI change, but it is simple to test: we can uncomment sdio define on pins file and test if file is filled.

dozer123456, 2020-07-30:
@davidtgbe will check later 👍

dozer123456, 2020-07-30:
@davidtgbe Not working in SDIO either PLR file is generated but is BLANK again.

Moody66, 2020-07-30:
Three prints. All printed 100%.   Nice work.

davidtgbe, 2020-07-30:
@dozer123456 ok. I will take a look at the code tomorrow. It seems clear that trigger works, but, there are problems generating PRL file content under power outage conditions as you have tested.

@Moody66  good to know. I will wait a little more before closing the issue... Just in case

davidtgbe, 2020-07-31:
Seems not having enough energy to complete the process.

```
---------------------------------> Outage simulation
>>> m413 s
SENDING:M413 S
>>> m413
SENDING:M413
echo:Power-loss recovery ON
SENDING:M413 O
echo:Power-loss recovery ON
Started saving routineWrite Job Recovery Info...
valid_head:1 valid_foot:1
current_position: 0.00,0.00,250.00,0.00
zraise: 0.00
home_offset: 0.00,0.00,0.00
position_shift: 0.00,0.00,0.00
feedrate: 1500
target_temperature: 0
target_temperature_bed: 0
fan_speed: 0
leveling: 0 fade: 0.00
sd_filename:
sdpos: 0
print_job_elapsed: 0
---
Writing to file: /PLR
Power-loss file write success!.
Power-loss file close success!.
Finished saving routineRec. Fallo electrico
Error:Printer halted. kill() called!

---------------------------------> Real printing outage

>>> m413 s
SENDING:M413 S
>>> m413
SENDING:M413
echo:Power-loss recovery ON
Started saving routineWrite Job Recovery Info...
valid_head:2 valid_foot:2
current_position: 0.00,0.00,250.00,0.00
zraise: 0.00
home_offset: 0.00,0.00,0.00
position_shift: 0.00,0.00,0.00
feedrate: 1500
target_temperature: 0
target_temperature_bed: 0
fan_speed: 0
leveling: 0 fade: 0.00
sd_filename:
sdpos: 0
print_job_elapsed: 0
---

```

dozer123456, 2020-07-31:
Any ideas on a solution? Could it write the data to eeprom ?

anigmad, 2020-07-31:
Is anet willing to share how they handled this?

davidtgbe, 2020-07-31:
I have to do some more tests... I have prioritize heaters shutdown code to be the first thing to be executed, maybe we could shutdown another thing or reorder code. Anyway, we always have had powerloss working without using the pin, so if it is a must have feature, you can work it out by commenting the powerloss pin definition. Only caution is taking care of SD health. 

@dozer123456 I think to have read somewhere that writing to eeprom is a marlin upcoming feature and It is not implemented right now.

@anigmad I don't know how Anet is dealing with it. May be Anet FW is more optimized in this sense...

Moody66, 2020-08-01:
Thats the problem with Marlin 2.. On 32 bit boards it uses a LOT more space.  Could even be part of the SD issue. I have read that same reason has caused issues with sd printing.

davidtgbe, 2020-08-01:
@Moody66 What do you mean? PLR info file size?

Moody66, 2020-08-01:
> @Moody66 What do you mean? PLR info file size?

https://all3dp.com/2/marlin-2-0-vs-marlin-1-differences/   Explain it well enough I guess.  There is another site that talks about SD with not a lot of memory. I will find it and post shortly    "

Despite all the improvements, Marlin 2 does have one large issue: memory consumption. When using Marlin 1, an 8-bit board’s memory allows for use of SD card support, auto-bed leveling, BLTouch support, and power loss recovery.

With Marlin 2, if you’re using an 8-bit board with the standard 128 KB of memory, you’re likely to run out. (Many standard printers, including the Ender 3, come with this board.) To enjoy all of Marlin 2’s features with an 8-bit board, you might have to disable SD card support and print on other platforms, like OctoPrint.

This fact leads many Marlin 2 users to consider upgrading to a 32-bit board with the standard 256 KB of memory. At the same time, there are some users who report that running Marlin 2 on an 8-bit board works fine; it’ll just depend on your printer and board.""

davidtgbe, 2020-08-02:
@Moody66  IMHO is not a matter fo space. ET4 mcu has enough ram, flash, CPU etc to handle everything. Every feature added to marlin does increase the final .bin file size. And, it has to fit in flash, but, in this case, we have plenty of room (1MB) to store the firmware 2 times.

I think it is a matter of firmware, drivers and stm32 framework maturity. We are working on bugfix 2.0.x brunch (not a stable release branch) with stm32 framework/HAL, and drivers, which are, in some cases, not fully tested or polished.

Moody66, 2020-08-04:
Ya. done a lot of reading up last couple of days on the hardware. Seems pretty solid. They were talking about ram running short after boot. 1meg is plenty though. I run Linux on a pi with 1m. So I am sure that cant be it now. Hopefully as the development improves it will help. Thanks for all you have done.

thinkyhead, 2020-08-30:
Stability is slowly improving for STM32 frameworks and Marlin HAL, and we're at a point now of doing some cleanup.

The power-loss recovery should be functioning better, if not perfectly, in the latest (2.0.6.1) release. As always, the feature is kind of unreliable unless there's a real power-loss-detection circuit, since it tends to re-start from the beginning of the current layer and get clogged. Writing more often can help, but then the card gets worn more quickly.

The part of Anet's unpublished ET4 firmware that most interests me is the code to update the firmware from the SD card. We've got something like that in Marlin now for some boards (with SDIO) but not this particular one. Maybe there's a way to remote debug the old `et4.bin` firmware as it's running on the board and see what kind of special bit-bashing it's doing to make the "Upload" feature work.

davidtgbe, 2020-09-09:
@thinkyhead I've merged latest bugfix source code, but powerloss doesn't work either. I thinks it is a matter of  the shutdown process order (optimization) or insufficient battery capacity to perform the process. Powerloss itself does work, but ir doesn't get completed running on battery. Anyway, it is not a very important feauture.

Regarding update process... Anet update process is not the best. If your FW is corrupt and you flash it, you lose your ability to reflash again. Bootloader by itself can't flash the firmware, it needs the FW assistance. Many users have had to buy a hardware flasher only to repair the board.

SidDrP, 2020-09-30:
Hello there @davidtgbe I have been closely following your et4 marlin port. Thanks to your effort i managed everything to work without issue.
I would like to add some info on PLR function in et4. Hope it helps.
Power loss pin PA8 remains at 3.3v but goes rapidly to 1.8v or less on power failure.  Immediately Pin26 PA3 goes high(please re check) triggering mosfet through diode d14 (marked yellow) keeping everything alive by super cap. And there's enough juice to store progress data. Btw in stock version eeprom is used. 
[image]
Led will glow till everything is energized.

davidtgbe, 2020-09-30:
@SidDrP nice analysis. Do you mean it is working for you oob with latest sources? I haven't tried since my last post in this thread.

SidDrP, 2020-09-30:
The hardware works but i don't have coding skills to make it working in marlin. I can do hardware stuff in my free time but software in nono. :(

thinkyhead, 2020-10-04:
I've just about got an ET4 patch ready, so it should be fully supported for Marlin 2.0.7.1. If anyone has well-tuned configurations feel free to submit your improved settings over at the Configurations repository.

davidtgbe, 2020-11-01:
Tried again today. Still not working and several issues regarding power loss are opened.

SidDrP, 2020-11-08:
@davidtgbe Can you please share how you have implemented the plr here and whats the issue. Did you used pin 26 to keep circuit energized. Thanks.

davidtgbe, 2020-11-08:
@SidDrP I have nothing to do here. Neither with software (marlin) nor with hardware (anet). Sorry.

SidDrP, 2020-11-08:
@davidtgbe I did some experiments with plr with almost success last weekend (eeprom save is still work in progress). I didn't followed through due to shortage of time. Also i don't think plr is useful for me as i do most prints in ABS on glass bed. Thought you might like to know but as you said "I have nothing to do here." so thanks for your time.

davidtgbe, 2020-11-08:
@SidDrP I do not use PLR either, but I am to help and of course I am interested in any solution to the problem, like any other user of the ET. My intention was to convey that I have not implemented the part of the software or the hardware, and that, to be honest, I have not done many tests. I have done nothing more than define the plr pin, as you would have read through this thread. That's all. If I can help you in something you just have to say it.

Zhiniukas, 2020-12-22:
I got PLR working on my Anet ET4+. 
Changes made:
In file pins_ET4.h pin for backup super capacitor is defined:   #define POWER_LOSS_PIN2                   PA3

In file powerloss.cpp, changes are here:
https://github.com/Zhiniukas/Marlin/blob/bugfix-2.0.x/Marlin/src/feature/powerloss.cpp

Now PLR works OK. On power loss plr file is created on SD card and printing resumes after the power is back. Or plr file is deleted if Stop print is selected from the menu.

davidtgbe, 2020-12-22:
@Zhiniukas Great! So there are 2 changes there:
- PA3 pin (as @SidDrP commented) to turn on SuperCap after a power outage.
- An SDCard bug fix (release and remount)
I will try these fixes, and, will merge those changes. Or if you want, do a PR (as you prefer).

Zhiniukas, 2020-12-22:
@davidtgbe Yes, you are right.
I made 2 major changes as you noticed. Also I changed resume procedure, allowing hot end heating before priming the extruder.

Please do changes yourself, as I am new to github. :)

davidtgbe, 2020-12-23:
@Zhiniukas Yes I also noticed you have swapped the hotend heating and volumetric extrusion state block code. 
However... taking a look at code and Marlin help, it seems setting volumetric state doesn't prime the extruder (it is a M200 command), but I have not tried by myself...
Purge seems to be performed on line commented as "//Additional purge if configured" related with parameter POWER_LOSS_PURGE_LEN

Are you sure this change is mandatory?

Zhiniukas, 2020-12-24:
@davidtgbe I checked my code again.
The change for heaters before volumetric extruder is not mandatory. Leave it as it is in original version.

Differences in my working code - I do not use   "#ifdef POWER_LOSS_PIN2" before setting pin2 high. I just set pin2 to desired state. Comment this check and test power loss. Also if you comment this line 
OUT_WRITE(POWER_LOSS_PIN2, LOW);
 you should see "Printer halted power loss" warning on each power off. If not, POWER_LOSS_PIN2 does not activate properly on power loss.

Also I added this line to marlincore.cpp just to set pin2 LOW on startup. Maybe it can be moved to pins_et4 file:

  #if ENABLED(POWER_LOSS_PIN)
    OUT_WRITE(POWER_LOSS_PIN2, LOW);                    //Power loss cap Off on startup.
  #endif

During the debugging of SD card access, I replaced src/sd card folder files with official Marlin 2.0.7.2 version files. But I did it only after debugging of print resume.

fizi1234, 2020-12-25:
work well wtih this Changes made:
In file pins_ET4.h pin for backup super capacitor is defined: #define POWER_LOSS_PIN2 PA3

In file powerloss.cpp, changes are here:
https://github.com/Zhiniukas/Marlin/blob/bugfix-2.0.x/Marlin/src/feature/powerloss.cpp

davidtgbe, 2020-12-30:
@Zhiniukas, 
Sorry, I haven't had much time to test it again. I just ran a couple of tests and it seems to be going very well. Very good work. I think we can close this issue.

closed 2020-12-30 by davidtgbe