davidtgbe/Marlin #4: Touch screen not functioning

dozer123456, 2020-08-09:
### Description

Since latest update, touch screen calibration is off

### Steps to Reproduce

1. [Switch on]
2. [touch the screen]

**Expected behavior:** [Corect function selected]

**Actual behavior:** [Unresponsive or incorrect function selected]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Configuration ET4 PRO.zip](https://github.com/davidtgbe/Marlin/files/5047500/Configuration.ET4.PRO.zip)

Moody66, 2020-08-09:
> ### Description
> Since latest update, touch screen calibration is off
>
> ### Steps to Reproduce
> 1. [Switch on]
> 2. [touch the screen]
>
> **Expected behavior:** [Corect function selected]
>
> **Actual [...]

Have you ran the calibration built into marlin????

dozer123456, 2020-08-09:
@Moody66 No. How do you get to the setting, not that I could get to it anyway as when I click on the screen it is to far off.

Can you run a G code command to initialize it?

dozer123456, 2020-08-09:
Sorted.  hold your finger on the screen for a few seconds opens up the Touch screen calibration settings.
🙈🙈🙈🙈🙈🙈🙈🙈🙈🙈🙈🙈🙈

dozer123456, 2020-08-09:
^Needs a M500 to save the settings after a power cycle.

It is strange as it worked by default before the last set of changes.

I have noticed that there is now no orientation in the config.h file??? 
I will add  " #define XPT2046_ORIENTATION TOUCH_PORTRAIT" to line 2216 and see if this sorts it.

dozer123456, 2020-08-09:
@davidtgbe 
adding " #define XPT2046_ORIENTATION TOUCH_PORTRAIT" to line 2216 in the config.h fixes this

davidtgbe, 2020-08-09:
Ok, thanks @dozer123456

davidtgbe, 2020-08-09:
@doezer123456 I've added the line. Not tested, but I've done a fast search through code and couldn't find any line in code referring to XPT2046_ORIENTATION. Does this line really works for you?

dozer123456, 2020-08-09:
Yes 🤔

dozer123456, 2020-08-09:
@davidtgbe 
[image]

davidtgbe, 2020-08-10:
@dozer123456  yes, I think I know what is happening. Latest merge from bugfix-2.0.0x changed some definitions, and, given de logic in:
https://github.com/davidtgbe/Marlin/blob/d0e728769568e2cedadcdae5c24a5a4a0a842f8d/Marlin/src/lcd/tft/touch.h#L49
I thought the line for TOUCH_ORIENTATION was not needed/used.
I think that, as TOUCH_ORIENTATION has changed from PORTRAIT to LANDSCAPE and, therefore, X and Y calibration parameters has to be swapped.
If you run a calibration by pressing touch by 2 or 3 seconds, calibration data regenerates for the new orientation, and that's why it works. 
That's my guess, but can't test it right now... Calibration data is printed on PC screen (i.e. pronterface) when you perform a manual calibration touching/holding screen by 2 or 3 seconds.

dozer123456, 2020-08-10:
@davidtgbe you are correct.

" #define XPT2046_ORIENTATION TOUCH_PORTRAIT" didn't actually work. I forgot to run a M502 to reset the EEPROM.

" X and Y calibration parameters has to be swapped." in the config.h file 
I tried this and it DID NOT WORK

If you make the highlighted changes to the Marlin/Marlin/src/lcd/tft/touch.h

#if !(defined(TOUCH_CALIBRATION_X) || defined(TOUCH_CALIBRATION_Y) || defined(TOUCH_OFFSET_X) || defined(TOUCH_OFFSET_Y) || defined(TOUCH_ORIENTATION))
  #if defined(XPT2046_X_CALIBRATION) && defined(XPT2046_Y_CALIBRATION) && defined(XPT2046_X_OFFSET) && defined(XPT2046_Y_OFFSET) **&& defined(XPT2046_ORIENTATION)**
    #define TOUCH_CALIBRATION_X  XPT2046_X_CALIBRATION
    #define TOUCH_CALIBRATION_Y  XPT2046_Y_CALIBRATION
    #define TOUCH_OFFSET_X       XPT2046_X_OFFSET
    #define TOUCH_OFFSET_Y       XPT2046_Y_OFFSET
    #define TOUCH_ORIENTATION    **XPT2046_ORIENTATION** 

And keep the " #define XPT2046_ORIENTATION TOUCH_PORTRAIT" on line 2216 in the config.h

this fixes the problem.

davidtgbe, 2020-08-10:
@dozer123456 dozer123456 that's a good fix. Tests correct. Commiting changes.

closed 2020-08-10 by davidtgbe