davidtgbe/Marlin #10: Anet ET4Pro configuration

Sebazzz, 2020-11-01:
I'm trying to compile for the ET4Pro - but I probably overlooking it - but what configuration would I use? 

It is mentioned in the README.md, and I know it uses TMC2208 drivers, but what about the other pinouts and bed leveling?

davidtgbe, 2020-11-01:
@Sebazzz you will have to write your own config. There are some configuration files on config folder, but, they are user configs and, they also are a outdated. You could also try easyconfig branch of this repo, which has a easyconfig.h file in which you can configure a working Marlin for your ET4 in a simple way. It is currently being tested, so you could givi it a go.

closed 2020-11-01 by davidtgbe

Sebazzz, 2020-11-01:
Apologies, I didn't notice the user configs folder.

Sebazzz, 2020-11-03:
@davidtgbe I got it working using the ezconfig branch, thanks!

One question: I noticed that according to the README there is an EEPROM on the board. Were you ever able to get it activated or is there no Marlin support for this particular EEPROM?

davidtgbe, 2020-11-03:
You are welcome. 
Yes there is. STM32 driver for the chip was not usable when I started this project. Maybe it is time to take it up again.