davidtgbe/Marlin #12: Power Loses recovery dont work on ET5 PRO

muhla89, 2021-01-07:
hello, i still have a problem with the bugfix. The problem is that in the event of a power failure, the printing no longer starts despite the setting being activated in the confic adv. I have already switched from false to true and also excluded the ARC but still no function does someone still have the same problem or the solution for it?

I did it according to these instructions

https://youtu.be/wlt7LIxvMJo

davidtgbe, 2021-01-08:
@muhla89 could you please try with @Zhiniukas fork? https://github.com/Zhiniukas/Marlin/
I have merged his changes, but try it and tell me if it works for you.

muhla89, 2021-01-08:
I have found the Problem

davidtgbe, 2021-01-08:
@muhla89 is it a problem with your printer or with the code. Could you detail your problem/solution?

muhla89, 2021-01-08:
With the Code 

the problem was that after a power failure the printing no longer went on now, after adjusting the code, it no longer does it and then continues printing where it left off

davidtgbe, 2021-01-08:
@muhla89 Could you make a PR or share your fix with us?

muhla89, 2021-01-09:
Yes thats the fix

davidtgbe, 2021-01-09:
@muhla89 which fix? You said that after adjusting the code it works for you. Could you be a little more detailed on the steps you have performed to fix your problem?

muhla89, 2021-01-09:
is difficult to describe, I looked for the values ​​from several pieces of information and adjusted them then it still didn't work because he had an error with powerloss with the UltraLCD then I told him that he should get the data from the marlinui.h and that's it it went

> Am 09.01.2021 um 10:39 schrieb David <notifications@github.com>:
> 
> @muhla89 which fix? You said that after adjusting the code it works for you. Could you be a little more detailed on the steps you have performed to fix your problem

davidtgbe, 2021-01-09:
@muhla89 ok. Then you can do this:
-ZIP the whole marlin folder you have fixed and upload the Zip file to https://wetransfer.com/upload for example, and share here the link or share it with me through telegram. My user is @davidtgbe.
-Tell me if you have fixed code on top of Zhiniukas fork or directly to my fork

Thanks

davidtgbe, 2021-01-11:
@muhla89 should I keep the issue open until you post your code or may be more info or should I close it?

closed 2021-01-12 by davidtgbe