davidtgbe/Marlin #17: Z-Offset no funciona

tutulino, 2021-04-13:
Hola, ante todo agradecerte el trabajo realizado y un tirón de orejas a Anet por cutres y rastreros....

Acabo de compilar y instalar una versión con bootloader incluida, todo parece funcionar bien (a falta de mas pruebas) excepto que el Z-offset no afecta a la altura de impresión, da igual con que valor empiezas la impresión, siempre comienza a la misma altura.
Si se activa el babystepping este si que afecta a la altura de la impresión en curso, pero si ese mismo valor lo pones en el offset, ya sea manualmente o porqué has establecido al compilar que se añada sólo, la siguiente impresión comienza a la altura "cero offset".
Voy a probar a poner el sensor de fin de carrera en la parte inferior a ver si pudiera estar relacionado con eso

tutulino, 2021-04-14:
Añado información: he probado a cambiar el sensor del eje Z a la parte inferior y ocurre lo mismo, la impresión no comienza con el offset asignado.

davidtgbe, 2021-04-26:
Hola @tutulino,
no sé que decirte, yo lo estoy usando sin problema. Usa el M851 y asegúrate de guardar el valor con un M500. Puedes comprobar si se aplica con un M503.
https://github.com/davidtgbe/Marlin/blob/d936018fcafef1dcc0103224e9e4e03cf16606c2/Marlin/EasyConfig.h#L158
Prueba a ver y me comentas, ya que soy consciente de que se usa y nadie más ha reportado fallos.

tutulino, 2021-04-26:
¿Una duda, la tuya es ET4 o ET4 pro? ¿con sensor de calibrado original o con bltouch?.

He probado de todas las maneras posibles y aunque el valor se guarda y la siguiente vez incluso en el menú partes de esa altura a la hora de la verdad empieza siempre a imprimir a la misma altura.

Creo que esta relacionado con el NOZZLE_AS_PROBE y que marlin entiende que si la sonda es la boquilla, no tiene sentido aplicar offset....pero es solo una sospecha.

davidtgbe, 2021-04-26:
Yo tengo la ET4 con placa silenciosa + BMG + BLTOUCH. El NOZZLE_TO_PROBE_OFFSET no tiene sentido si tu sonda es la de la ET4 normal. No hay offset, ya que la sonda y el nozzle están en la misma posición. El M851 en tu caso no sirve para nada.
Que valor es el que estás cambiando? De todas formas, pásate por el Telegram de la ET4. Allí hay gente que te echará una mano en tiempo real. Yo últimamente no estoy mucho por allí, pero si me mencionan suelo responder.

tutulino, 2021-04-26:
Con la sonda original no hay offset en X e Y como dices pero si en Z ya que hay el offset que provoca la propia sonda que suele ser muy poco, pero algo es. Antes de abrir la incidencia estuve buscando por el grupo de Telegram, pero lo único que encontré era gente que le pasaba lo mismo, consultaban y nadie respondía.

El problema es con el M851, el babystep funciona bien si lo activas y se acumula al offset si se lo indicas en el firmware, peo a la siguiente impresión el offset guardado no se usa.

Estoy bastante convencido de que a nadie con la ET4 y sonda original le debe funcionar el offset guardado, mas que nada porque aunque no lo he encontrado escrito me huele a que al activar el NOZZLE_AS_PROBE ignora el NOZZLE_TO_PROBE_OFFSET en los tres ejes, cuando en este caso para que fuese funcional debería hacerlo sólo en X e Y.

davidtgbe, 2021-04-27:
Puede que sea así. Prueba a definirlo como sonda capacitiva (fix_mounted_probe) o si usas easyconfig ET_LV_MOUNTED_PROBE. A fin de cuentas, el pin es el mismo. Deja los offset X e Y a 0 y varia el Z a ver si te funciona asi.

closed 2021-05-07 by davidtgbe