davidtgbe/Marlin #20: et5 update

manotroll, 2021-06-26:
this marlin is being updated with the main one?
or can i get the settings here and use in the main marlin?

davidtgbe, 2021-06-26:
I "regularly" merge main branch changes in this fork (using bugfix 2.0.x). But, I only do it if some interesting feature (from my pov) has been  added). You can take what you want from this fork, and do whatever mix you want. That's what opensource is about.

I don't get you with the warning about the board build offset, sorry. Could you be more detailed?

manotroll, 2021-06-26:
in platformio.ini file does not have offset to allow constant update by openblt
I don't know how to take a part of the code and use it in the official version and I don't know what changes from there to here either.

davidtgbe, 2021-06-26:
@manotroll, platform.ini has been split some time ago. Read point 1.3 of the read.me carefully:

```
If you are going to take option A from considerations section (not using a BL), you can skip this step. 
Otherwise, you need to offset the firmware to give some room to the BL. 
That is achieved by uncommenting a line in \ini\stm32f4.ini file. 
It is commented by default
```

If you don't know how to do it, just stick with one of the marlin forks (this, or another) or use the main marlin branch. I can't help you here with that, sorry.

manotroll, 2021-06-27:
it's elsewhere in the tutorials I've seen don't have this comment

davidtgbe, 2021-06-27:
I think readme.md is the only place where bootloader instructions are described. But, if you point me to the place you have seen it, I will be pleased to correct it.

manotroll, 2021-06-27:
another doubt I saw that you commented that the example files are not valid do you know where I can find these files?

davidtgbe, 2021-06-27:
If you mean where the path to those outdated files are: https://github.com/davidtgbe/Marlin/tree/bugfix-2.0.x/config/users%20configs
If you are looking for help on configuring your config files,  try on marlin forums and web page, anet FB groups, or telegram: https://t.me/anetet4esp

Just google it.

closed 2021-06-27 by davidtgbe