davidtgbe/Marlin #24: G29, M48 and Easyconfig not working

santiis2010, 2021-08-30:
### Description

### Steps to Reproduce

1. Dowload Marling 2.0 bug fix for Anet ET5, configure easyconfig.h and compile without errors
2. Flashing the ET5 printer with silent board, flash process its ok
3. Doing G28 works perfect, then G29 stops at first probe.

**Expected behavior: With G29 it has to do all the 9 probe points, motor drivers to be TCM2208 stand alone

**Actual behavior: Printer stops at first probe point, some times at the 6 probe point with error "Probe out of bounds", motor drivers shown as regular A4988

#### Additional Information
- I tried to configurate configuration.h at the same time using easyconfig.h to see if that can bee fixed.

Printer was working just fine, i needed to update it because i fixed the leng of the 
* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

[Marlin.zip](https://github.com/davidtgbe/Marlin/files/7078952/Marlin.zip)

davidtgbe, 2021-09-01:
Hello Santiago,
should be working fine. If you edit with VS CODE Marlin\src\inc\EasyConfigPost.h, you can check which parameters are currently used (light font style) and unused (shadow/greyed out font style) as in the screen capture below using your config files:
https://github.com/davidtgbe/Marlin/blob/a6c0bcb42a9162d8f63363420b17a42b6700fbd7/Marlin/src/inc/EasyConfigPost.h

[image]

As you can see, TMC2208_STANDALONE drivers are assigned.

I have noticed you have changed some parameters in configuration.h, which, easyconfig will just get overrided. You can find more info in comments at the end of the easyconfig.h file:
https://github.com/davidtgbe/Marlin/blob/a6c0bcb42a9162d8f63363420b17a42b6700fbd7/Marlin/EasyConfig.h#L185-L280

For example, you can't change print area by modifying size parameters in configuration.h file if you use easyconfig.h, because easyconfig will just overwrite these settings here:
https://github.com/davidtgbe/Marlin/blob/a6c0bcb42a9162d8f63363420b17a42b6700fbd7/Marlin/src/inc/EasyConfigPost.h#L281-L321

I recommend you to do this test to isolate problems:
- Redownload this repo
- Uncomment the line ```#define ET_EASY_CONFIG``` on configuration.h
- Paste your custom easyconfig.h file (or modify it according to your printer) and change the ```#define ET_CUSTOM_MACHINE_NAME``` to some custom name like "ET5 SANTIAGO TMC2208" which let you know that easyconfig has taken effect.
- Build and flash
- Follow post-installation steps below: https://github.com/davidtgbe/Marlin/blob/a6c0bcb42a9162d8f63363420b17a42b6700fbd7/Marlin/EasyConfig.h#L137-L164

Let me know if drivers are assigned correctly and M503 is reflecting your changes.

santiis2010, 2021-09-01:
> Hello Santiago,
> should be working fine. If you edit with VS CODE Marlin\src\inc\EasyConfigPost.h, you can check which parameters are currently used (light font style) and unused (shadow/greyed out [...]

So i did al this, the only things i changed on easyconfigpost.h was the offset of the bed and X axis (X-3 and Y-22) and the steps from E0 to 98..68 i moved the extruder to direct drive.

(its ok if i can write in spanish?)

So i compiled everything, formated a 8GB card (not the one that came with the printer), inserted the micro SD card on the printer, so BTL started to update the firmware.

I connect the printer with windows 10, and pronterface, i send m502 and then m500, set the M851 X-24 Y-8.7 and Z0, pids are just fine, everything worked like a charm till now... So i send G28 and on the LCD screen for the printer while doing home i can see that X and Y gives me this numbres while moving the axis "X-529 and Y-483 (wierd) but when finish the homing it says X-3 and Y-22.

Then i send G29, it goes to the middle, just fine, then when its going to do the firs probe point 1/9 the Z axis on the lcd screen says Z-2 and the printer stop doing ABL, on Pronterface o get this error message:

SENDING:G29
echo:busy: processing
echo:busy: processing
//action:notification Sondeo Fallado
Error:Probing Failed
[ERROR] Error:Probing Failed

//action:notification ET5 Silenciosa Santiago Lista
ok P7 B2

Also when i do M48 the first time before i sent the G29 the probe works but on the LCD screen Z axis says Z-2, so the probe its doing auto leveling and M48 at z-2 or something like that.

I attached againg the new files.
[Marlin.zip](https://github.com/davidtgbe/Marlin/files/7092793/Marlin.zip)

davidtgbe, 2021-09-01:
> So i compiled everything, formated a 8GB card (not the one that came with the printer), inserted the micro SD card on the printer, so BTL started to update the firmware.

OK

>  connect the printer with windows 10, and pronterface, i send m502 and then m500, set the M851 X-24 Y-8.7 and Z0, pids are just fine, everything worked like a charm till now... 

Aquí un par de cosas:
- M502 para borrar, después el M851 para establecer la distancia entre la punta de sondeo y la punta del hotend y por último, el M500 para guardar, ese es el orden. 
- Si has puesto Z0 es que tienes la sonda a la misma altura que el hotend? Esto me parece raro... No es lo normal. Lo normal es tener la sonda por encima del hotend. Puedes ver este video para usar el asistente de offset en Z. https://youtu.be/fN_ndWvXGBQ?t=411

> So i send G28 and on the LCD screen for the printer while doing home i can see that X and Y gives me this numbres while moving the axis "X-529 and Y-483 (wierd) but when finish the homing it says X-3 and Y-22.

Con el primer home, lo importante es que se inicialicen y se hagan corresponder el home de X, Y, y Z con el offset marcado (-3 y -22). Si al finalizar el home es lo que te marca, es correcto.

> Then i send G29, it goes to the middle, just fine, then when its going to do the firs probe point 1/9 the Z axis on the lcd screen says Z-2 and the printer stop doing ABL, on Pronterface o get this [...]

Veo que el nombre de la impresora te lo está recogiendo correctamente del easyconfig. Entiendo que el resto de parámetros también (como los drivers que comentabas al principio). 
Veo también el error, pero es poco detallado. Llegados a este punto, si con lo anterior no se te soluciona, una prueba más que puedes hacer es usar bilinear (es lo que yo uso y no tengo pegas), aunque no creo que sea ese el problema.

> Also when i do M48 the first time before i sent the G29 the probe works but on the LCD screen Z axis says Z-2, so the probe its doing auto leveling and M48 at z-2 or something like that.

En posición de Home, Z debe marcar el offset que le hayas puesto tú con el M851. Si marca Z=-2 teniendo puesto Z=0 con el M851, algo te falla.
Comprueba con un M503 siempre después de cambiar algo y salvarlo con M500.

Envía una captura del M503 si después de todo te sigue sin funcionar y adjunta también los cambios que haces en el easyconfigpost.h
Hablamos en Español, no hay problema

davidtgbe, 2021-09-10:
Hola Santiago, al final, diste con el problema? Si es así, por saberlo y cerrar el hilo, y sino, para intentar solventar el bug entre los dos.

santiis2010, 2021-09-10:
> Hola Santiago, al final, diste con el problema? Si es así, por saberlo y cerrar el hilo, y sino, para intentar solventar el bug entre los dos.

Hola David, justo te iba a responder y vi mensaje, lleve la impresora al Service de Anet en Uruguay, resulta que en un momento el sensor ya ni funcionaba, me dijeron que el sensor está roto, y también se "jodió" algo en la placa porque no detecta si le pones un sensor nuevo, de momento le tendré que poner un endstop, vos sabés algún esquema donde pueda ver qué componentes en la placa manejan el sensor para intentar reparar? 

En todo caso esto está resuelto ya que es tema de hardware y no de software ni firmware.

davidtgbe, 2021-09-11:
Siento lo de la placa... 
Que yo sepa no hay un esquema tan detallado Santiago. Tendrías que seguir tu mismo las pistas en el PCB e ir comprobando que todo está en orden con ayuda de un multimetro.

closed 2021-09-11 by davidtgbe

santiis2010, 2021-09-13:
Si sabes que seguí las pistas pero es raro, el hotend y la cama me dan
temperaturas de 58 y 62 grados como de ambiente, por lo tanto debe estar
jodida alguna resistencia, y algún smd del sensor de nivelación, hablé con
anet y hay escasez de chips 😕 por lo que no tienen placas silenciosas, voy
a ver si escribiéndoles a ellos me pasarían un esquema porque las pistas no
están muy claras en la olaca, igual no creo que me pasen pero bueno.

Gracias por la ayuda capo! Que andes bien