davidtgbe/Marlin #33: Missing #define SD_SS_PIN in pins_ANET_ET4.h

ricmm71, 2023-03-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

There is no explicit define for SD_SS_PIN in file pins_ANET_ET4.h, so it is assigned the value PIN_SPI_SS in file Marlin\src\HAL\STM32\spi_pins.h , and PIN_SPI_SS is assigned to PA4 in file buildroot\share\PlatformIO\variants\MARLIN_F4x7Vx\variant.h .
When function spiBegin(void) from file Marlin\src\HAL\STM32\HAL_SPI.cpp is called (when accesing SD card), the pin PA4 is written with a 1. But this pin is the BED thermistor pin, so it should never be written, only read.
The SD_SS_PIN definition should be added in file pins_ANET_ET4.h:
[...]
  #if DISABLED(SDIO_SUPPORT)
    #define SOFTWARE_SPI
    #define SDSS                     SDIO_D3_PIN
    **#define SD_SS_PIN                SDSS**
    #define SD_SCK_PIN               SDIO_CK_PIN
    #define SD_MISO_PIN              SDIO_D0_PIN
    #define SD_MOSI_PIN              SDIO_CMD_PIN
  #endif
[...]

### Version of Marlin Firmware

Marlin bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2023-03-11:
This will be fixed upstream: https://github.com/MarlinFirmware/Marlin/pull/25492

davidtgbe, 2023-03-14:
Merged upstream, removed SD_SS. I think only SD_SS_PIN is used . Should work, but let me know if not.

closed 2023-03-14 by davidtgbe

Talixme, 2024-01-27:
Muy buenas;

Te escribo en este issues porque no me deja crear uno nuevo.

Muchas gracias por este gran proyecto y soporte que le das, es lo que hizo que me comprara esta impresora.

El año pasado instalé Marlin con tu tutorial mediante el método de Openblt con la idea de ir actualizando si tener que usar el ordenador y el lector de tarjetas no funcionaba, pero como lo tenía con una raspberry y Octoprint, pues no me importaba demasiado, pero hoy por distintos errores necesitaba imprimir por SD, así que ahora a la 01:00 me dispongo a actualizar el firmware a la última versión viendo (sin yo recordar si antes lo soportaba) que ya soporta SD. Descargo , Modifico Easyconfig, ET4_Plus, compilo flasheo por SD ( que si deteca Openblt) y nada el mismo error en la pantalla "Fallo al iniciar SD"

He probado distintas tarjetas y nada, vamos de hecho la misma tarjeta si la lee una Anet A8 sin problema también con Marlin

No sé si estaré haciendo algo mal o cuál es el problema, pero si me puedes echar una mano ya la dejo actualizada y funcionando al 100%

Mil gracias y un saludo.