davidtgbe/Marlin #35: EasyConfig plugin [BUG] (bug summary)

manotroll, 2023-09-24:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

EasyConfig configuration via vscode is not applying the changes

### Bug Timeline

EasyConfig plugin

### Version of Marlin Firmware

2.0.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

EasyConfig configuration via vscode is not applying the changes

davidtgbe, 2023-10-29:
@manotroll, you forgot attaching configuration files when you opened the issue.
Did you uncomment easyconfig line on configuration.h? VScode (platform.ide) is applying correctly easyconfig settings as far as I know.

closed 2023-10-29 by davidtgbe

manotroll, 2023-10-30:
https://i.imgur.com/JcO1RKq.png
but there would be no way to add it and an error in vs studio