davidtgbe/Marlin #36: [BUG] config/users configs/ET4/davidtgbe_bltouch/Configuration.h

manotroll, 2023-10-30:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

when using this file it is giving an error defining the TMC2208 silent board but it is giving random errors requesting changes

### Version of Marlin Firmware

Marlin-bugfix-2.0.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/davidtgbe/Marlin/files/13199815/Marlin.zip)

manotroll, 2023-10-30:
https://github.com/davidtgbe/Marlin/tree/bugfix-2.0.x/config/users%20configs/ET4/davidtgbe_bltouch

davidtgbe, 2023-11-17:
Please, read documentation.
https://github.com/davidtgbe/Marlin/tree/bugfix-2.1.x/config

b) Configure your build from scratch.
[User configurations](https://github.com/davidtgbe/Marlin/blob/bugfix-2.1.x/config) are neither updated nor maintaned. Please, read the readme.md inside the config folder. Don't copy paste these files.
[Marlin configurations](https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/Anet) can also be a place to take a look. Don't copy paste these files as there could be a version mismatch between them and this repo.

closed 2023-11-17 by davidtgbe